DHDAXCW/OpenWRT_x86_x64 #9: kmod-ixgbevf 安装问题

mk-marc, 2022-03-02:
大佬，请问这个插件怎样才能装上？是10G NIC的SV-IOV驱动。目前得到的的提示：

Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Unknown package 'kmod-ixgbevf'.

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.102-1-0b4ec6aebc8126ed102c57f2d6545d61) for kmod-mdio
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.102-1-0b4ec6aebc8126ed102c57f2d6545d61) for kmod-pps
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.102-1-0b4ec6aebc8126ed102c57f2d6545d61) for kmod-ptp
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.102-1-0b4ec6aebc8126ed102c57f2d6545d61) for kmod-hwmon-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.102-1-0b4ec6aebc8126ed102c57f2d6545d61) for kmod-libphy
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.102-1-0b4ec6aebc8126ed102c57f2d6545d61) for kmod-mdio-devres
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.102-1-0b4ec6aebc8126ed102c57f2d6545d61) for kmod-ixgbe
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.102-1-0b4ec6aebc8126ed102c57f2d6545d61) for kmod-ixgbevf
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ixgbevf found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-ixgbevf.

DHDAXCW, 2022-03-11:
你先把上面依赖安装全了，可以再安装你的驱动 依赖如下：
`
kmod-mdio
kmod-pps
kmod-ptp
kmod-hwmon-core
kmod-libphy
kmod-mdio-devres
kmod-ixgbe
kmod-ixgbevf
`

shiyipeng210, 2022-04-18:
```
$` opkg install kmod-mdio kmod-pps kmod-ptp kmod-hwmon-core kmod-libphy kmod-mdio-devres kmod-ixgbe kmod-ixgbevf
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package kmod-mdio (5.15.33-1) installed in root is up to date.
Package kmod-pps (5.15.33-1) installed in root is up to date.
Package kmod-ptp (5.15.33-1) installed in root is up to date.
Package kmod-hwmon-core (5.15.33-1) installed in root is up to date.
Package kmod-libphy (5.15.33-1) installed in root is up to date.
Package kmod-mdio-devres (5.15.33-1) installed in root is up to date.
Package kmod-ixgbe (5.15.33-1) installed in root is up to date.
Unknown package 'kmod-ixgbevf'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.111-1-758c474763454be8ccb8c15838343cc3) for kmod-mdio
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.111-1-758c474763454be8ccb8c15838343cc3) for kmod-pps
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.111-1-758c474763454be8ccb8c15838343cc3) for kmod-ptp
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.111-1-758c474763454be8ccb8c15838343cc3) for kmod-hwmon-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.111-1-758c474763454be8ccb8c15838343cc3) for kmod-libphy
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.111-1-758c474763454be8ccb8c15838343cc3) for kmod-mdio-devres
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.111-1-758c474763454be8ccb8c15838343cc3) for kmod-ixgbe
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.111-1-758c474763454be8ccb8c15838343cc3) for kmod-ixgbevf
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ixgbevf found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-ixgbevf.
```

大佬，我试了还是不行

closed 2022-07-19 by DHDAXCW