DHDAXCW/OpenWRT_x86_x64 #37: 最新版总是重启，内存4G 60G固态 J1900

RmondJone, 2022-11-03:
最新版总是重启，内存4G 60G固态 J1900，之前是因为不能定时重启，然后换了一个最新版，结果刷了，重启个不停，没有设置定时任务。上个版本就是不能重启，但是很稳定不会掉线重启，这个最新版重启个不停，一天能重启5 6次，我绑了阿里云动态DNS，一天收到变更通知5 6次邮件

RmondJone, 2022-11-03:
[image]

DHDAXCW, 2022-11-05:
等换个内核

xt101, 2022-11-05:
已回5.15.72版本

RmondJone, 2022-11-06:
> 等换个内核

换完了通知一下，我再刷新下

DHDAXCW, 2022-11-06:
我在5.15.75-76上遇到过 77就好了，今天重新编译上传固件

DHDAXCW, 2022-11-06:
如果再重启我立马锤了它，改5.15.72~

xt101, 2022-11-06:
感谢大佬，回去刷一下77版本

RmondJone, 2022-11-06:
> 如果再重启我立马锤了它，改5.15.72~

大佬牛逼，哈哈哈

xt101, 2022-11-06:
> 我在5.15.75-76上遇到过 77就好了，今天重新编译上传固件

大佬，74版本可以用？

RmondJone, 2022-11-07:
[image]
用了77版本内核，已经稳定运行14个小时，再观察看看。话说大佬为啥又把77版本下了？？

xt101, 2022-11-08:
刷了最新的77版本，重启后网络显示连接，但是上不了网，后台也进不了，回退到72版本了

DHDAXCW, 2022-11-08:
> 刷了最新的77版本，重启后网络显示连接，但是上不了网，后台也进不了。回退到72版本，点重启后能很快重新连网。

你是畅网吗？是的话 直接连eth1进后台

xt101, 2022-11-08:
> > 刷了最新的77版本，重启后网络显示连接，但是上不了网，后台也进不了。回退到72版本，点重启后能很快重新连网。
> 
> 你是畅网吗？是的话 直接连eth1进后台

大佬您好，主机是intel nuc。

DHDAXCW, 2022-11-08:
> > > 刷了最新的77版本，重启后网络显示连接，但是上不了网，后台也进不了。回退到72版本，点重启后能很快重新连网。
> > 
> > 
> > 你是畅网吗？是的话 直接连eth1进后台
> 
> 大佬您好，主机是intel nuc。可能重启后接口换了是吧

我先测试最新，上午改了几个东西

RmondJone, 2022-11-08:
> 刷了最新的77版本，重启后网络显示连接，但是上不了网，后台也进不了。回退到72版本，点重启后能很快重新连网。

我这边没出现这种情况，77版内核挺稳的

xt101, 2022-11-09:
> > > > 刷了最新的77版本，重启后网络显示连接，但是上不了网，后台也进不了。回退到72版本，点重启后能很快重新连网。
> > > 
> > > 
> > > 你是畅网吗？是的话 直接连eth1进后台
> > 
> > 
> > 大佬您好，主机是intel nuc。可能重启后接口换了是吧
> 
> 我先测试最新，上午改了几个东西

最新版通过备份升级可以吗，之前的77版本升级没成功。

DHDAXCW, 2022-11-09:
解压

xt101, 2022-11-09:
> 解压
了解

xt101, 2022-11-15:
最新版的6.0.8固件装在m2 ssd上启动重启更快了

RmondJone, 2022-11-15:
> 最新版的6.0.8固件装在m2 ssd上启动重启更快了

固件版本都出到6.0.8了？这么快，我用5.14.77挺稳的，稳定才是最重要的

DHDAXCW, 2022-11-15:
我发现只要别升级，这就正常很久了~~~之前我一直升级后 老是重启，搞得快烦死了。然后我全新刷入就正常，也不重启、。、

RmondJone, 2022-11-17:
[image]

14号刷的77版本，貌似还是不稳，2天重启一次

RmondJone, 2022-11-17:
都是格式化硬盘，重新刷的，是什么原因导致重启的？日志能查到原因吗？怎么看？

RmondJone, 2022-11-17:
我在想会不会是什么定时任务啊？固件里有定时任务吗？我路由器设置的周六5点重启一次，但是目前没有按照这个规则走。我再观察观察几天看看，看看重启有没有规律

xt101, 2022-11-17:
> 我发现只要别升级，这就正常很久了~~~之前我一直升级后 老是重启，搞得快烦死了。然后我全新刷入就正常，也不重启、。、

是的，全新刷入是最稳妥。

xt101, 2022-11-17:
> 解压

大佬，6.0.8固件会使usb网卡掉线，需要加kmod-usb3驱动

RmondJone, 2022-11-17:
哎，还是不行，夜里莫名其妙的又重启2次
[image]

RmondJone, 2022-11-20:
刷了作者的2个版本都会不定时重启，我已经放弃了，自己编译了一个自用的版本，有需要的同学可以下载使用，只编了x86，其他架构没编译，有需要的同学自取。
[https://github.com/RmondJone/lede/releases/tag/1.1.0](https://github.com/RmondJone/lede/releases/tag/1.1.0)

closed 2022-11-20 by RmondJone

RmondJone, 2022-11-20:
我初步怀疑总是重启的原因，是因为插件太多，然后内核区间只分配64M导致的。建议作者分区的时候，内核空间分大一点，我自制的内核空间是512M，已经用了几天没出现过这种情况

yiyule10, 2022-11-23:
> 我初步怀疑总是重启的原因，是因为插件太多，然后内核区间只分配64M导致的。建议作者分区的时候，内核空间分大一点，我自制的内核空间是512M，已经用了几天没出现过这种情况

可以说一个内核空间的大小在哪里改吗？

RmondJone, 2022-11-23:
> 

刷我的固件，我自己编译了一个，你要是X86架构的就能用，我分配了512M内核空间，1G软件包空间

yiyule10, 2022-11-23:
> > 
> 
> 刷我的固件，我自己编译了一个，你要是X86架构的就能用，我分配了512M内核空间，1G软件包空间

好的，谢谢。但是，我还是想知道在哪里修改，比较好奇，可以说一个吗？

RmondJone, 2022-11-23:
> > > 
> > 
> > 
> > 刷我的固件，我自己编译了一个，你要是X86架构的就能用，我分配了512M内核空间，1G软件包空间
> 
> 好的，谢谢。但是，我还是想知道在哪里修改，比较好奇，可以说一个吗？

只有在编译固件的时候才能指定，已经打出来的固件，改不了这个。另外Overlay也是一样的，虽然有方法去改，但都不是真正意思的去改分区大小。都是先把原来的废了，然后重新挂载一个更大的分区。但是原有的分区就会永久的在那边占用，浪费硬盘空间。最好的方法就是会自己编译固件，想改多大改多大。也不难，自己网上找找教程，就是编译比较耗时。短则2小时，长则1天。

yiyule10, 2022-11-23:
> > > > 
> > > 
> > > 
> > > 刷我的固件，我自己编译了一个，你要是X86架构的就能用，我分配了512M内核空间，1G软件包空间
> > 
> > 
> > 好的，谢谢。但是，我还是想知道在哪里修改，比较好奇，可以说一个吗？
> 
> 只有在编译固件的时候才能指定，已经打出来的固件，改不了这个。另外Overlay也是一样的，虽然有方法去改，但都不是真正意思的去改分区大小。都是先把原来的废了，然后重新挂载一个更大的分区。但是原有的分区就会永久的在那边占用，浪费硬盘空间。最好的方法就是会自己编译固件，想改多大改多大。也不难，自己网上找找教程，就是编译比较耗时。短则2小时，长则1天。

好的，明白了，感谢指点。