DHDAXCW/OpenWRT_x86_x64 #48: Transmission web is not accessible

godpusa, 2023-02-27:
version: 2023.02.22 x86-64-Linux-6.1.12
Sorry, but there is no log with transmission.

I can not access the web interface of transmission.
it said:
"
404: Not Found

Couldn't find Transmission's web interface files!

Users: to tell Transmission where to look, set the TRANSMISSION_WEB_HOME environment variable to the folder where the web interface's index.html is located.

Package Builders: to set a custom default at compile time, #define PACKAGE_DATA_DIR in libtransmission/platform.c or tweak tr_getClutchDir () by hand.
"
it seems no transmission-web has been installed.

$ opkg list-installed | grep transmission
luci-app-transmission - git-23.048.51027-d67627a-1
luci-i18n-transmission-ca - git-23.048.51027-d67627a-1
luci-i18n-transmission-cs - git-23.048.51027-d67627a-1
luci-i18n-transmission-de - git-23.048.51027-d67627a-1
luci-i18n-transmission-el - git-23.048.51027-d67627a-1
luci-i18n-transmission-es - git-23.048.51027-d67627a-1
luci-i18n-transmission-fr - git-23.048.51027-d67627a-1
luci-i18n-transmission-he - git-23.048.51027-d67627a-1
luci-i18n-transmission-hu - git-23.048.51027-d67627a-1
luci-i18n-transmission-it - git-23.048.51027-d67627a-1
luci-i18n-transmission-ja - git-23.048.51027-d67627a-1
luci-i18n-transmission-ms - git-23.048.51027-d67627a-1
luci-i18n-transmission-no - git-23.048.51027-d67627a-1
luci-i18n-transmission-pl - git-23.048.51027-d67627a-1
luci-i18n-transmission-pt - git-23.048.51027-d67627a-1
luci-i18n-transmission-pt-br - git-23.048.51027-d67627a-1
luci-i18n-transmission-ro - git-23.048.51027-d67627a-1
luci-i18n-transmission-ru - git-23.048.51027-d67627a-1
luci-i18n-transmission-sk - git-23.048.51027-d67627a-1
luci-i18n-transmission-sv - git-23.048.51027-d67627a-1
luci-i18n-transmission-tr - git-23.048.51027-d67627a-1
luci-i18n-transmission-uk - git-23.048.51027-d67627a-1
luci-i18n-transmission-vi - git-23.048.51027-d67627a-1
luci-i18n-transmission-zh-cn - git-23.048.51027-d67627a-1
luci-i18n-transmission-zh-tw - git-23.048.51027-d67627a-1
transmission-daemon-openssl - 2.94-8
transmission-web-control - 2020-09-13-c26a0761-1

zJanch, 2023-03-23:
直接卸载吧，用不了，换一个

Jazzxianren, 2023-07-19:
I met same issue in version: 2023.06.19 x86-64-Linux-5.15.116 and version: 2023.07.19 x86-64-6.1.38.

closed 2024-01-05 by DHDAXCW