DHDAXCW/OpenWRT_x86_x64 #50: 2023.02.22 x86-64-Linux-6.1.12 固件 不定时异常重启

how0723, 2023-03-07:
2023.02.22 x86-64-Linux-6.1.12 固件 docker版本 不定时异常重启，基本上每天一次，时间不固定

how0723, 2023-03-07:
Tue Mar  7 04:06:59 2023 daemon.info avahi-daemon[3878]: New relevant interface eth1.IPv4 for mDNS.
Tue Mar  7 04:06:59 2023 daemon.info avahi-daemon[3878]: Registering new address record for 192.168.1.9 on eth1.IPv4.
Tue Mar  7 04:06:59 2023 daemon.notice netifd: Interface 'wan' is now up
Tue Mar  7 04:06:59 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Tue Mar  7 04:07:00 2023 daemon.info avahi-daemon[3878]: Joining mDNS multicast group on interface br-lan.IPv6 with address fe80::528d:5cff:feab:d98c.
Tue Mar  7 04:07:00 2023 daemon.info avahi-daemon[3878]: New relevant interface br-lan.IPv6 for mDNS.
Tue Mar  7 04:07:00 2023 daemon.info avahi-daemon[3878]: Registering new address record for fe80::528d:5cff:feab:d98c on br-lan.*.
Tue Mar  7 04:07:00 2023 user.notice firewall: Reloading firewall due to ifup of wan (eth1)
Tue Mar  7 04:07:00 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (eth1)
Tue Mar  7 04:07:00 2023 cron.err crond[7826]: crond (busybox 1.36.0) started, log level 5
Tue Mar  7 04:07:01 2023 daemon.notice procd: /etc/rc.d/S91wrtbwmon: Warning: using existing /tmp/usage.db
Tue Mar  7 04:07:01 2023 daemon.err usbmuxd[8673]: [04:07:01.548][3] usbmuxd v1.1.1 starting up
Tue Mar  7 04:07:01 2023 daemon.err usbmuxd[8673]: [04:07:01.548][3] Using libusb 1.0.26
Tue Mar  7 04:07:01 2023 daemon.err usbmuxd[8673]: [04:07:01.549][3] Initialization complete
Tue Mar  7 04:07:01 2023 daemon.err usbmuxd[8673]: [04:07:01.549][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Tue Mar  7 04:07:01 2023 daemon.info rclone: Instance "rclone" is disabled.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Leaving mDNS multicast group on interface eth1.IPv6 with address fe80::2e0:4cff:fe68:6500.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Joining mDNS multicast group on interface eth1.IPv6 with address 240e:36f:d54:77f0::2.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Registering new address record for 240e:36f:d54:77f0::2 on eth1.*.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Withdrawing address record for fe80::2e0:4cff:fe68:6500 on eth1.
Tue Mar  7 04:07:02 2023 daemon.notice netifd: Interface 'wan6' is now up
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Registering new address record for 240e:36f:d54:77f0:2e0:4cff:fe68:6500 on eth1.*.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Leaving mDNS multicast group on interface br-lan.IPv6 with address fe80::528d:5cff:feab:d98c.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Joining mDNS multicast group on interface br-lan.IPv6 with address 240e:36f:d54:77f2::1.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Registering new address record for 240e:36f:d54:77f2::1 on br-lan.*.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Withdrawing address record for fe80::528d:5cff:feab:d98c on br-lan.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Withdrawing address record for 192.168.2.1 on br-lan.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Withdrawing address record for 240e:36f:d54:77f0:2e0:4cff:fe68:6500 on eth1.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Withdrawing address record for 240e:36f:d54:77f0::2 on eth1.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Withdrawing address record for 192.168.1.9 on eth1.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Withdrawing address record for ::1 on lo.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Withdrawing address record for 127.0.0.1 on lo.
Tue Mar  7 04:07:02 2023 daemon.warn avahi-daemon[3878]: Host name conflict, retrying with OpenWrt-2
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Registering new address record for 240e:36f:d54:77f2::1 on br-lan.*.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Registering new address record for 192.168.2.1 on br-lan.IPv4.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Registering new address record for 240e:36f:d54:77f0:2e0:4cff:fe68:6500 on eth1.*.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Registering new address record for 240e:36f:d54:77f0::2 on eth1.*.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Registering new address record for 192.168.1.9 on eth1.IPv4.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Registering new address record for ::1 on lo.*.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Registering new address record for 127.0.0.1 on lo.IPv4.
Tue Mar  7 04:07:02 2023 daemon.info avahi-daemon[3878]: Registering HINFO record with values 'X86_64'/'LINUX'.
Tue Mar  7 04:07:02 2023 user.info mwan3rtmon[6587]: Detect rtchange event.
Tue Mar  7 04:07:03 2023 daemon.err nginx[9098]: 2023/03/07 04:07:03 [emerg] 9098#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Tue Mar  7 04:07:03 2023 daemon.err nginx[9098]: 2023/03/07 04:07:03 [emerg] 9098#0: bind() to [::]:443 failed (98: Address in use)
Tue Mar  7 04:07:03 2023 daemon.err nginx[9098]: 2023/03/07 04:07:03 [emerg] 9098#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Tue Mar  7 04:07:03 2023 daemon.err nginx[9098]: 2023/03/07 04:07:03 [emerg] 9098#0: bind() to [::]:80 failed (98: Address in use)
Tue Mar  7 04:07:03 2023 daemon.err nginx[9098]: 2023/03/07 04:07:03 [emerg] 9098#0: bind() to 0.0.0.0:443 failed (98: Address in use)
[... cut ...]
Tue Mar  7 07:34:37 2023 daemon.info netdata[16664]: Initializing file /var/cache/netdata/ip.tcpconnaborts/TCPAbortFailed.db.
Tue Mar  7 08:43:49 2023 daemon.info netdata[16664]: Initializing file /var/cache/netdata/ipv4.tcperrors/main.db.
Tue Mar  7 08:43:49 2023 daemon.info netdata[16664]: Initializing file /var/cache/netdata/ipv4.tcperrors/InErrs.db.
Tue Mar  7 08:43:49 2023 daemon.info netdata[16664]: Initializing file /var/cache/netdata/ipv4.tcperrors/InCsumErrors.db.
Tue Mar  7 08:43:49 2023 daemon.info netdata[16664]: Initializing file /var/cache/netdata/ipv4.tcperrors/RetransSegs.db.

johnsmith8736, 2023-03-07:
> 2023.02.22 x86-64-Linux-6.1.12 固件 docker版本 不定时异常重启，基本上每天一次，时间不固定

我也遇到一样的问题！我现在换成3.5的了

johnsmith8736, 2023-03-07:
> 2023.02.22 x86-64-Linux-6.1.12 固件 docker版本 不定时异常重启，基本上每天一次，时间不固定

再观察下会不会也不定时重启！

how0723, 2023-03-07:
每天至少一次重启，不定时的

johnsmith8736, 2023-03-07:
> 每天至少一次重启，不定时的

是的…我以前刷过的固件也出现这种情况！应该是固件的问题！换了就好了

johnsmith8736, 2023-03-07:
2023.03.05 x86-64-Linux-5.15.98 固件也有不定时重启的毛病！哎～…

how0723, 2023-03-07:
感觉固件里面是不是加了私货啊

godpusa, 2023-03-07:
我用的是2.22的full版本，没有重启问题。

how0723, 2023-03-07:
请问怎样抓取重启前的日志？ 怎么排查原因？ @johnsmith8736   @godpusa

johnsmith8736, 2023-03-07:
> 请问怎样抓取重启前的日志？ 怎么排查原因？ @johnsmith8736 @godpusa

不知道…

how0723, 2023-03-08:
Wed Mar  8 02:56:22 2023 daemon.crit mdadm[2116]: DeviceDisappeared event detected on md device /dev/md0
Wed Mar  8 02:56:22 2023 daemon.notice procd: /etc/rc.d/S13openssl: Generating engines.cnf
Wed Mar  8 02:56:22 2023 daemon.notice procd: /etc/rc.d/S13openssl: Enabling engine devcrypto
Wed Mar  8 02:56:23 2023 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Mar  8 02:56:23 2023 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Mar  8 02:56:23 2023 user.notice : Added device handler type: 8021ad
Wed Mar  8 02:56:23 2023 user.notice : Added device handler type: 8021q
Wed Mar  8 02:56:23 2023 user.notice : Added device handler type: macvlan
Wed Mar  8 02:56:23 2023 user.notice : Added device handler type: veth
Wed Mar  8 02:56:23 2023 user.notice : Added device handler type: bridge
Wed Mar  8 02:56:23 2023 user.notice : Added device handler type: Network device
Wed Mar  8 02:56:23 2023 user.notice : Added device handler type: tunnel
Wed Mar  8 02:56:23 2023 daemon.notice procd: /etc/rc.d/S45nfs: exportfs: could not open /var/lib/nfs/.etab.lock for locking: errno 2 (No such file or directory)
Wed Mar  8 02:56:23 2023 daemon.notice procd: /etc/rc.d/S45nfs: exportfs: could not open /var/lib/nfs/.etab.lock for locking: errno 2 (No such file or directory)
Wed Mar  8 02:56:23 2023 daemon.notice procd: /etc/rc.d/S45nfs: exportfs: can't lock /var/lib/nfs/etab for writing
Wed Mar  8 02:56:23 2023 daemon.notice procd: /etc/rc.d/S50autoreboot: Command failed: ubus call service delete { "name": "cron" } (Not found)
Wed Mar  8 02:56:23 2023 daemon.notice procd: /etc/rc.d/S50autoreboot: Auto REBOOT has started.
Wed Mar  8 02:56:23 2023 cron.err crond[2953]: crond (busybox 1.36.0) started, log level 8
Wed Mar  8 02:56:24 2023 kern.info kernel: [   18.302343] eth0: 0xffffc9000002d000, 50:8d:5c:ab:d9:8c, IRQ 124
Wed Mar  8 02:56:24 2023 kern.info kernel: [   18.369565] br-lan: port 1(eth0) entered blocking state
Wed Mar  8 02:56:24 2023 kern.info kernel: [   18.374928] br-lan: port 1(eth0) entered disabled state
Wed Mar  8 02:56:24 2023 daemon.notice netifd: Interface 'lan' is enabled
Wed Mar  8 02:56:24 2023 daemon.notice netifd: Interface 'lan' is setting up now
Wed Mar  8 02:56:24 2023 daemon.notice netifd: Interface 'lan' is now up
Wed Mar  8 02:56:24 2023 daemon.notice netifd: Interface 'loopback' is enabled
Wed Mar  8 02:56:24 2023 daemon.notice netifd: Interface 'loopback' is setting up now
Wed Mar  8 02:56:24 2023 daemon.notice netifd: Interface 'loopback' is now up
Wed Mar  8 02:56:24 2023 kern.info kernel: [   18.380364] device eth0 entered promiscuous mode
Wed Mar  8 02:56:24 2023 kern.info kernel: [   18.386998] eth1: 0xffffc90000075000, 00:e0:4c:68:65:00, IRQ 125
Wed Mar  8 02:56:24 2023 daemon.notice netifd: Interface 'wan' is enabled
Wed Mar  8 02:56:24 2023 daemon.notice netifd: Interface 'wan6' is enabled
Wed Mar  8 02:56:24 2023 daemon.notice netifd: Network device 'lo' link is up
Wed Mar  8 02:56:24 2023 daemon.notice netifd: Interface 'loopback' has link connectivity
Wed Mar  8 02:56:25 2023 user.notice mwan3[3079]: Using firewall mask 0x3F00
Wed Mar  8 02:56:25 2023 user.notice mwan3[3079]: Max interface count is 60
Wed Mar  8 02:56:25 2023 daemon.err odhcpd[2831]: Failed to send to ff02::1%lan@br-lan (Permission denied)
Wed Mar  8 02:56:25 2023 authpriv.info dropbear[3942]: Not backgrounding
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: Found user 'nobody' (UID 65534) and group 'nogroup' (GID 65534).
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: Successfully dropped root privileges.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: avahi-daemon 0.8 starting up.
Wed Mar  8 02:56:25 2023 daemon.warn avahi-daemon[3947]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: No service file found in /etc/avahi/services.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: Joining mDNS multicast group on interface br-lan.IPv4 with address 192.168.2.1.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: New relevant interface br-lan.IPv4 for mDNS.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: Joining mDNS multicast group on interface lo.IPv6 with address ::1.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: New relevant interface lo.IPv6 for mDNS.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: Joining mDNS multicast group on interface lo.IPv4 with address 127.0.0.1.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: New relevant interface lo.IPv4 for mDNS.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: Network interface enumeration completed.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: Registering new address record for 192.168.2.1 on br-lan.IPv4.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: Registering new address record for ::1 on lo.*.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: Registering new address record for 127.0.0.1 on lo.IPv4.
Wed Mar  8 02:56:25 2023 daemon.info avahi-daemon[3947]: Registering HINFO record with values 'X86_64'/'LINUX'.
Wed Mar  8 02:56:25 2023 daemon.info nginx_init: using /etc/nginx/uci.conf (the test is ok)
Wed Mar  8 02:56:25 2023 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/dhcp
[... cut ...]
Wed Mar  8 02:56:52 2023 daemon.info netdata[16497]: NETDATA_SYSTEM_VIRTUALIZATION=none
Wed Mar  8 02:56:52 2023 daemon.info netdata[16497]: NETDATA_SYSTEM_VIRT_DETECTION=none
Wed Mar  8 02:56:52 2023 daemon.info netdata[16497]: NETDATA_SYSTEM_CONTAINER=unknown
Wed Mar  8 02:56:52 2023 daemon.info netdata[16497]: NETDATA_SYSTEM_CONTAINER_DETECTION=none
Wed Mar  8 02:56:52 2023 daemon.info netdata[16497]: NETDATA_SYSTEM_CPU_LOGICAL_CPU_COUNT=4
Wed Mar  8 02:56:52 2023 daemon.info netdata[16497]: NETDATA_SYSTEM_CPU_VENDOR=GenuineIntel

djy0716, 2023-03-12:
我也有同样的问题

3.11的非docker full固件也有这个问题

closed 2023-03-12 by DHDAXCW