DHDAXCW/OpenWRT_x86_x64 #74: docker貌似出了些小问题

lemontealin, 2024-04-06:
不影响使用的小问题，本来应该在电报群说的，结果绝了...号被电报封了，正在申诉😂

系统重启后进管理页面点docker栏就会显示下图这样，十几秒到半分钟左右会恢复正常，感觉是启动卡了一下，之前版本没遇到过，不知道会不会有其他影响。

[image]

部分日志
[image]

[image]

lemontealin, 2024-04-06:
一觉醒来被电报的反垃圾邮件系统ban了，莫名其妙，还在申诉，现在电报群和私聊都不能发消息😂

ohyeah521, 2024-04-06:
路由器不玩docker保平安

lemontealin, 2024-04-06:
有没有一种可能，提供上网基础服务有硬路由，软路由只是提供扩展功能...

DHDAXCW, 2024-04-06:
等我把电报机器人给日了

DHDAXCW, 2024-04-06:
改好了，等晚上再试试

lemontealin, 2024-04-06:
感谢骷髅佬！😘❤️

closed 2024-04-06 by lemontealin