DHDAXCW/OpenWRT_x86_x64 #77: 后台更新失败

abcang0323, 2024-04-25:
您好，

我是小白一枚，在openwrt界久仰您的大名，感谢您的付出。
从418版开始使用您的固件，感觉清爽简洁，所有常用的功能基本全都包括了，非常喜欢。423更新后尝试后台更新，但两次都失败了。不知道我的方法是都正确，下载最新固件-系统-备份更新-上传固件-刷写，同样的方法两次失败，解压后也试了一下，同样也失败了，麻烦您指教一下，毕竟不希望每次都重新安装的，感谢🙏。

DHDAXCW, 2024-04-26:
内存要4GB以上

abcang0323, 2024-04-26:
好的。我回去试一下。硬盘不需要调整吧？谢谢您🙏

abcang0323, 2024-04-26:
还有一个问题想请教您一下，想导入ssl证书进行安全访问，发现系统已经安装了ssl插件，但找不到uhttpd，我应该怎么做呢？搜索了很久一直没找到合适的答案，麻烦您了。

abcang0323, 2024-04-26:
> 内存要4GB以上

将内存扩容到4G后升级两次未成功，解压后再试，成功了。

tom5920, 2024-04-30:
[image]
2G一样流畅

abcang0323, 2024-05-04:
> 内存要4GB以上
您好，上次增加内存到4G后台更新后，系统运行正常，但内存占用很高，不知道是否正常。
[image]

IngwerHe, 2024-05-08:
作者能不能帮忙把hello world加进去呀？passwall用不习惯

DHDAXCW, 2024-05-10:
> 作者能不能帮忙把hello world加进去呀？passwall用不习惯

helloworld早跑了怎么加进去？

closed 2024-05-12 by abcang0323