DHDAXCW/OpenWRT_x86_x64 #104: 2024.12.03固件ssh无法访问，动态ddns报错

rayle2, 2024-12-09:
1.远程ssh连接不上，只能在ttyd终端里连接

2.ddns运行报错，提示：
 085621       : verbose mode  : 0 - run normal, NO console output
 085621  WARN : No update_url found/defined or no update_script found/defined! - TERMINATE
 085621  WARN : PID '11430' exit WITH ERROR '1' at 2024-12-09 08:56

rayle2, 2024-12-10:
> 1.远程ssh连接不上，只能在ttyd终端里连接
> 
> 2.ddns运行报错，提示： 085621 : verbose mode : 0 - run normal, NO console output 085621 WARN : No update_url found/defined or no update_script found/defined! - TERMINATE 085621 WARN : PID '11430' exit WITH ERROR '1' at 2024-12-09 08:56

ddns提供商是dnspod，其他未测

LemonLatteFizz, 2024-12-14:
ssh问题没遇到
自带的ddns我也不能用，11月的版本开始就是，猜测可能换了新版ddns或者上游固件的问题，用ddns-go吧