DHDAXCW/OpenWRT_x86_x64 #110: 好像samba4不太正常

LemonLatteFizz, 2024-12-24:
连接不上，不知道为啥，似乎是启动失败。明天研究研究

icarlhsu, 2024-12-25:
我用的ARM版，删了ksmbd后samba4就能用了，但是重启op后samba4不能自启动。可以上恩山论坛看看

LemonLatteFizz, 2024-12-25:
> 我用的ARM版，删了ksmbd后samba4就能用了，但是重启op后samba4不能自启动。可以上恩山论坛看看

感谢感谢，能用了
软件包里删除ksmbd-server立马就能用了，应该就是没删掉这个的原因 @DHDAXCW 
不过跟你一样，重启之后samba4不能自启动，我先加了个启动脚本，剩下晚上再看看

LemonLatteFizz, 2024-12-25:
> 我用的ARM版，删了ksmbd后samba4就能用了，但是重启op后samba4不能自启动。可以上恩山论坛看看

我好像搞定了，可以自启了
vi /etc/init.d/samba4 把优先级98改成99
然后/etc/init.d/samba4 enable，重启op

不过不确定是改优先级有用还是启动服务命令有用。如果是优先级问题，那可能是samba4启动时候盘还没挂载？总之现在是可以自启了

DHDAXCW, 2024-12-25:
> > 我用的ARM版，删了ksmbd后samba4就能用了，但是重启op后samba4不能自启动。可以上恩山论坛看看
> 
> 感谢感谢，能用了 软件包里删除ksmbd-server立马就能用了，应该就是没删掉这个的原因 @DHDAXCW 不过跟你一样，重启之后samba4不能自启动，我先加了个启动脚本，剩下晚上再看看

ok 下版本移除

DHDAXCW, 2024-12-25:
请问你刷的是最新的吗？

LemonLatteFizz, 2024-12-25:
> > > 我用的ARM版，删了ksmbd后samba4就能用了，但是重启op后samba4不能自启动。可以上恩山论坛看看
> > 
> > 
> > 感谢感谢，能用了 软件包里删除ksmbd-server立马就能用了，应该就是没删掉这个的原因 @DHDAXCW 不过跟你一样，重启之后samba4不能自启动，我先加了个启动脚本，剩下晚上再看看
> 
> ok 下版本移除

感谢大佬，还有个samba4不能开机自启的bug，目前通过改优先级能用

avijynisc2, 2025-01-06:
大佬，24版的软件包还是有问题，一直是 ：软件包列表已超过 24 小时未更新

LemonLatteFizz, 2025-01-06:
> 大佬，24版的软件包还是有问题，一直是 ：软件包列表已超过 24 小时未更新

这个正常，不用管，点更新列表不爆红就行

avijynisc2, 2025-01-06:
> > 大佬，24版的软件包还是有问题，一直是 ：软件包列表已超过 24 小时未更新
> 
> 这个正常，不用管，点更新列表不爆红就行

好的！谢谢。。。 这固件非常NS， 解决了我困扰已久的PD下发,

closed 2025-01-06 by LemonLatteFizz